intel/acat #2: ACAT Vision: "This app can't run on your PC"

mxstbr, 2015-08-17:
When I try starting ACAT Vision, I get an error saying "This app can't run on your PC". The rest of the ACAT suite runs perfectly fine and works.

Windows 10 PRO (32-bit) on a Toshiba Laptop.

naninga, 2015-08-21:
Same happens to me with a Toshiba Satellite L745 - and Widows 8, any solution will be great!

patzerr, 2015-08-21:
Hi,

since this Program seems to be written in C#, you need to install the .NET Runtime Environment.
See:

"It is written in C# using Microsoft Visual Studio 2012 and .NET 4.5 and runs on Microsoft Windows 7 or later"

Download:
https://msdn.microsoft.com/en-us/en-en/library/5a4x27ek%28v=vs.110%29.aspx

Or Install Visual Studio 2012, that should install .NET automatically.

hth

khanning, 2015-08-27:
Note 4 in the README may be relevant to this issue:
>The ACAT Vision software is 64-bit and will not run on 32-bit Windows systems. The ACAT application however is 32-bit and will run on both 32-bit and 64-bit systems.