intel/acat #4: Restarting the Presage engine

schacon, 2015-08-19:
Presage seems to die fairly frequently and I can't figure out how to restart the engine. If I run the installer again to uninstall and then reinstall it, it seems to start working again. Running the service tray program in the `bin` directory seems to put the icon in the service tray but ACAT won't connect to it. Is there a good way to manage the Presage engine?

takumikub, 2015-08-21:
Oooo

ClayF, 2015-08-23:
Uninstalling and running the application as an administrator fixed this problem for me. The repair/modify functionality of the installer did not fix it.

Crozix, 2015-08-24:
If ClayF's solution doesn't work it may help running "ACAT Vision" before starting the main program.

H1ghty, 2015-08-29:
Reinstalling doesn't work for me.