intel/acat #5: Fatal error. Error setting word prediction engine to [Presage Word Predictor]

chiragjindal, 2015-08-20:
installed on windows 10 pro

schacon, 2015-08-20:
This is actually the same as #4 - this is the message I get too. The Presage WCF service will run for a while and then die somehow and it does not seem possible to get it restarted without uninstalling and reinstalling the service entirely.

H1ghty, 2015-08-29:
Same problem here. Got Presage installed but ACAT can't see it.

nanomo, 2015-08-29:
@schacon you can start it from C:\Program Files (x86)\presage\bin\presage_wcf_service_system_tray.exe i think is not a service installed as a REAL service in windows becouse i dont see any name related in the list of services in taskmanager

arpitvkarma, 2016-07-12:
Same Issue "Fatal error. Error setting word prediction engine to [Presage Word Predictor]". Intel ACAT ran quite well for 1-2 week then suddenly one day this error starts popping up always.
OS: Windows 7 64 bit.
ACAT v0.99

wandick, 2016-08-16:
I solved this problem by installing Microsoft Visual C++.

mgalib, 2017-11-23:
Need help !!!

I have exactly same problem : "Fatal error. Error setting active word prediction engine"

I tried setting compatibility mode to Windows 7 for ACATapp.exe. I also tried installing again as admin. And I tried installing Microsoft Visual C++.

I have windows home 10. My computer is brand new so there is almost no other programs.

Please help .... I am stuck !!!!

treysears13, 2019-02-20:
I tried installing Microsoft Visual C++ and that didn't work I tried uninstalling then reinstalling the program and it still didn't work. I am  running out of ideas to fix this. Please help. Thanks.  `<?xml version="1.0"?>
<configuration>
  <startup useLegacyV2RuntimeActivationPolicy="true">
    
  <supportedRuntime version="v4.0" sku=".NETFramework,Version=v4.5"/></startup>
  <appSettings>

    <add key="SendKeys" value="SendInput"/>

  </appSettings>
  <system.serviceModel>
        <bindings>
              <netNamedPipeBinding>
                    <binding name="NetNamedPipeBinding_Presage" />
              </netNamedPipeBinding>
        </bindings>
        <client>
              <endpoint address="net.pipe://localhost/PresageService/v1/presage"
                    binding="netNamedPipeBinding" bindingConfiguration="NetNamedPipeBinding_Presage"
                    contract="PresageService.Presage" name="NetNamedPipeBinding_Presage">
              </endpoint>
        </client>
    </system.serviceModel>
</configuration>
I have left the code for the program exe attatched.

treysears13, 2019-02-20:
I solved my problem by going to the troubleshooting guide and thoroughly reading through it and finding out that when I installed ACAT presage was not installed because my account was not set as administrator of this computer I am installing presage now and if It works I will post again confirming whether it works or not.

wytzor, 2021-08-25:
HI, same issue for me Fatal error. Error setting active word prediction engine
I really need the software to work.
I tried setting compatibility mode to Windows 7 for ACATapp.exe. I also tried installing again as admin. And I tried installing Microsoft Visual C++.
Note: Installed ACATSetup.exe as an Administrator, then re-installed presage as an Administrator. Still not working.
One thing to note is that ACAT/Install contains presage-32bit whereas my System is 64 bit. Is this the cause of the problem?

Update:
Tried and installed presage (64-bit) still the same Issue "Fatal error. Error setting active word prediction engine". Tried running presage standalone it is working fine. I think the issue is at Intel ACAT side

Could someone help me please. I want to help my uncle to communicate with us 
Is there a community around this app?
I read the troubleshooting guide.

closed 2024-11-19 by csavur