intel/acat #6: Port to GNU/Linux platform request

ftoledo, 2015-08-21:
Just this. What about a GN/Linux port?

kevin-coyle, 2015-08-21:
I think with .NET being open sourced now this may actually be a lot easier than previous.

kevin-coyle, 2015-08-22:
The build errors when compiling on Xamarin on OSX with mono 4.0.1 are related to `System.Windows.Automation` not being found on my system. I think mono has that library. Maybe someone else can take a look too?

KaleidonKep99, 2015-08-22:
http://www.mono-project.com/archived/ui_automation/

pizzamaker, 2015-08-23:
This would be an excellent step forward.

H1ghty, 2015-08-29:
I think that support for Android would be cool ;)

alexandre-mbm, 2016-07-27:
The demonstration videos shows use for things such as Notepad, Windows Explorer and Internet Explorer. The "scanners" detects possible actions for those contexts. I do not see how this could be easely ported to Linux or Android.

xloem, 2021-07-19:
Still, doing this is so important.  If we work together, we can eventually make it happen.  Factoring logic away from implementation would be a great first step.  The https://mycroft.ai/ project has some similarities to this one.

xloem, 2024-11-20:
hey, it’s great to hear this was finally accomplished! would you be able to link to the work, commit, or instructions somewhere? or was this closed in error?

csavur, 2024-11-20:
Dear @xloem, this issue closed due to it was open 9 years ago. We are currently supporting only windows OS due to low level dependency to Windows OS. At this point we do not have a plan to support Linux or Mac. 

However, I will re-open this issue as it is still relevant to the current version as well.