intel/acat #10: acat_gestures.exe has stopped working

leticiayb, 2015-08-27:
Hello!

Everything is working fine, except on the ACAT Vision software. Everytime I try to open it, a message appears saying "acat_gestures.exe has stopped working". I've alredy followed the instructions and installed both VSU_4\vcredist_x64.exe and VSU_4\vcredist_x86.exe. For the fisrt one, the  "acat_gestures.exe has stopped working" error appears and for the second one, the "The program can't start because MSVCP110.DLL is missing from your computer" error appears. I don't know what to do anymore and I'll be very grateful if someone could help me! 

Thank you very much,
Leticia Botelho

willianfatec, 2015-08-27:
Leticia,

Just look if can you found that DLL in your computer. If you had find, just put it in the same executable's directory. If you google about it, do you see that is a common error in installation of games that need vcredist_x??.exe.

 Good luck!

nanomo, 2015-08-28:
@leticiayb ACAT needs Microsoft .NET Framework 4.5, try to check in your windows components if its installed, in theory ACAT will install if it not present.

closed 2024-11-19 by csavur