intel/acat #11: Localization.

ibakaidov, 2015-08-27:
Hi guys! I am russian programer. I create keyboard for disables, whoose control only one button. 
I wanna translete acat for russians. Can i do it?

nanomo, 2015-08-27:
the code must be restructured to fit language translation, they say it will be included in the future at the moment i had working the predictive text with spanish and will soon make a tutorial, but first you need a the database predictive on your language (like database.db, is  a sqllite).

in other hand, can i see how you modify the keyboards? im helping a person with movility problems and had some old keyboards to experiment.

molerat619, 2015-08-27:
How did u do it in spanish? Where did u get the database from?

ibakaidov, 2015-08-27:
Can you see https://github.com/JuliusSweetland/OptiKey/issues/73? I wrote about my DisQwerty then.

brlima94, 2015-08-28:
##### Aviso, versão em português disponível abaixo.

> *English Version:*

Hi,

On the recent Intel release on the opening of ACAT software source-code, we (a group of students of a Information Systems bachelor's degree from a São Paulo University) decided to translate the system interface into Portuguese as a product of our project .

We downloaded the source-code and started to research, but we would like to know if there is anyone else doing the translation into Portuguese and if we need to do any formal request in order to do so.

Our intent upon completion of the translation is that the ACAT software can be translated more quickly and conveniently to a larger number of languages, thus expanding the number of people who may use this tool in their native language.

We wonder if there is anyone who has begun this work to exchange experiences and avoid duplicate work, and we are looking for people interested in doing the translation to other languages.

Regards,
Bruno Lima.

> *Versão em Português:*

Olá,

Na recente publicação da Intel sobre a abertura do código-fonte do software ACAT, nós (um grupo de estudantes de um bacharelado em Sistemas de Informação de uma Universidade de São Paulo) decidimos traduzir a interface do sistema para o Português como um produto do nosso projeto.

Nós fizemos o download do código-fonte e começamos a pesquisa, mas gostaríamos de saber se há qualquer outra pessoa fazendo a tradução para o Português e se precisamos de qualquer pedido formal, a fim de fazê-lo.

Nossa intenção após a conclusão da tradução é que o software ACAT possa ser traduzido de forma mais rápida e convenientemente a um maior número de línguas, ampliando assim o número de pessoas que podem usar esta ferramenta em sua língua nativa.

Gostaríamos de saber se há alguém que começou este trabalho para trocar experiências e evitar trabalho duplicado, e estamos em busca de pessoas interessadas em fazer a tradução para outros idiomas.

Atenciosamente,
Bruno Lima.

nanomo, 2015-08-28:
@molerat619 install presage 0.9.1 and then in the path ```C:\Program Files (x86)\presage\share\presage\```  you will get a tiny database_es.db , database_en.db and database_de.db, matching witch predictive database for spanish, english and deutch language.

So copy database_es.db to your desktop, then rename it to database.db and copy and replace to:
C:\Intel\ACAT\Users\Default\WordPredictors\Presage
C:\Intel\ACAT\Install\Users\Default\WordPredictors\Presage

and rename it to database_en.db and copy and replace it to  
C:\Program Files (x86)\presage\share\presage\

this will let you use a spanish predictive text, the only issue are the latin chars but i will fix this soon :dancer:

molerat619, 2015-08-28:
Wow, cool, thank you! I will try that later today!
--
Best regards,
Marcel Izgin Freitag,

nanomo, 2015-08-28:
@brlima94 i think we all want, think the best way its to make a list of keywords in a string_en.resx for english once we had the complete list, we can translate it to our native language and make the pullrequest here or maybe work in some temporal repo for those files :D https://github.com/nanomo/acat-tmp

molerat619, 2015-08-28:
I did 'make install' on my MacBook and I got following Output. Where can I find the database now?

[image]

nanomo, 2015-08-28:
@molerat619 don't have any idea on macos try <presage_Install_path>/share/presage/

why not to use in windows 7 ? ACAT was made targeting that.

molerat619, 2015-08-28:
I'm not so familiar with compiling on windows, so I wanted to do the compiling part on mac. okay i will look for it and if I can't find it, i will try it on windows.

could you maybe upload those databases somewhere so that we can save future users the whole presage-part?

molerat619, 2015-08-29:
I've found it on Mac, but it only contains _en, _es and _it - no _de :/

H1ghty, 2015-08-29:
So for now Presage got predictive database for english, german and spanish language? or there are more to find?  I can't find folder like this C:\Program Files (x86)\presage\share\presage\ 

How this database looks like? it is possible to open it? maybe

nanomo, 2015-08-29:
@molerat619 sorry my bad :D, its as you say: _en,_es_it

nanomo, 2015-08-29:
@H1ghty the presage predictive databases are in SQLite, you can opened with http://sqlitebrowser.org/ wich works in windows and mac :)

H1ghty, 2015-08-29:
@nanomo  Yeah, already noticed that :) 
So as I saw Presage supports autocomplete for whole sentences, that's why there are tables with columns "word 1", "word 2", "word 2" I guess. I'm just wondering if it could be possible to just make it to support any language but only for single word autocomplete and just put words from a language dictionary to the database ;) and then adjust the app for that ;)

molerat619, 2015-08-29:
@nanomo too bad :D do you have an idea where else I can get it from?

robertorigotto, 2015-09-05:
Olá @brlima94 ! 

Sou desenvolvedor e tenho interesse em auxiliar na tradução do ACAT para Português (menus e Presage). Você já fez alguma evolução ?

Obrigado,
Roberto Rigotto

Tapan-Aggarwal, 2015-09-06:
Hi Everyone.Can anyone of you share me working copy of ACATsetup.exe on my mail id tapan.aggarwalse@gmail.com

H1ghty, 2015-10-17:
Hi,
I replaced database with polish language, but unfortunately I have no polish characters in running application. Anybody knows how to fix it?

I think that soon I will share with you all how to replace database, but first I will try to do as much as I can myself and then give more details ;)

H1ghty, 2015-11-01:
@nanomo Hi, did you fix the latin chars problem? Could you share how to fix it? I got that problem too but with polish chars. 

BR, H1ghty

A01221624, 2016-01-28:
Regarding the non-ASCII / Unicode characters: I haven't checked ACAT's code nor am I exactly sure of the issue you're running into, but I have a guess. Here's my take:

### Preamble
ACAT gets predictions from Presage. Presage is a C++-written library using std::string which, when consumed by C#-written ACAT, get encoded incorrectly (as far as I understand).

### Potential Solution
Recode Presage related strings before using them (and potentially also those fed back to Presage). https://stackoverflow.com/questions/14057434/how-can-i-transform-string-to-utf-8-in-c

Hopefully that's it... (@H1ghty)

~NaN

P.S. Regarding Presage's localized databases: The ones on the `presage\share\presage\` folder are just samples, they are going to give... unideal predictions if used directly; for example, the one for spanish is based on El Quijote de la Mancha, first published in 1605 (because it's now free, and of course they used a more recent edition, but still, imagine the language).

Then, you should use the text2ngram (found in `presage\bin\`) tool provided with Presage to generate a database more appropriate to your user(s). Just feed it with a ton of text written by the potential user(s) and have it compute 1-grams, 2-grams, and 3-grams (that means you run the tool three times with the same output file), and feed that file to ACAT through the corresponding XML configuration file (something like `C:\Intel\ACAT\Users\ACAT\PresageWordPredictorSettings.xml`). That way you can use any language and also have predictions that are best suited for the user :).

A better explanation on that by Presage's author http://sourceforge.net/p/presage/mailman/message/29275071/

alexandre-mbm, 2016-07-27:
@brlima94, meu pai sofre de ELA e eu tenho habilidades de programação. Estou interessado em ajudar nas traduções para Português do Braisl. Meu e-mail: alexandre.mbm@gmail.com.