intel/acat #12: ACATsetup.exe not found in downloaded acat-0.9.1.zip

Tapan-Aggarwal, 2015-08-30:
Your ACAT page in this repository is saying you would be releasing the new version of ACAT soon is that the reason the downloaded acat-0.9.1.zip is not having ACATsetup.exe file. If that is so how soon would intel put the ACATsetup.exe in the GITHUB repository for the us to download.
Thanks & regards,
Tapan Aggarwal
Delhi India

tarsiscr, 2015-09-02:
Hello, I need to try it.
Could u help me, sending me an email when the setup is avaliable?
Thks
Társis

nanomo, 2015-09-02:
they say it will be available later this week, stay in touch :)

niyazialpay, 2015-09-02:
I can't find too

Micheli123, 2015-09-05:
Im having the same problem i can not find the ACATsetup.exe file.

Please could you help us?

closed 2024-11-19 by csavur