intel/acat #14: New release of ACAT now available

saiprasadb01, 2015-09-08:
The binaries for ACAT v0.97.9.1 are now available <a href=https://github.com/01org/acat/releases>here</a>.  A number of new features have been added including a QWERTY Alphabet scanner layout, Talk applications, support for Windows Explorer, Outlook, Media Player.  
Please upgrade your current version.
Source code for this release will be uploaded soon.

mattengel, 2015-09-08:
The zip file that keeps appearing for download is called acat-master.zip.  It is very small (287KB) as opposed to the original download I picked up last month which was several MB in size.  Can you point me in the right direction to find the download?

Thanks!

nanomo, 2015-09-08:
@mattengel the zip doesnt have nothing than a pdf and some txt files, there isnt any source codes at the moment, you must download the installer https://github.com/01org/acat/releases/download/v0.97.9.1%60/ACATSetup.exe

mattengel, 2015-09-09:
Thank you very much! For the life of me, I couldn't seem to locate where
the exe file was located to download it.  Thanks for sending the link!  I'm
testing the software with different hardware for my brother who has ALS.
His ability to speak is nearly gone and we are looking for options to help
him communicate with us.

Thanks again!

Matt

kinderbyte, 2015-09-25:
Do you have a date for the source code? Thanks

closed 2024-11-19 by csavur