intel/acat #19: Need help in getting ACAT to run on Windows 7

mayankkh, 2015-10-22:
I have a friend who is suffering from ALS. I was able to get ACAT working on a Windows 8.1 laptop for her. However, I need to install it on another laptop which has Windows 7. I am able to successfully install it and run it on Windows 7. However, any gestures that I do on ACAT Vision are detected on the ACAT vision but are not sent to ACAT tryout.

Also when I exit ACAT tryout, ACAT vision doesn't exit. So I think ACAT tryout and ACAT vision are not able to communicate to each other on windows 7. But when I installed ACAT on Windows 8.1 then everything works.

@saiprasadb01 or anyone could someone please suggest on how to get it running on Windows 7.

The windows 7 laptop is 64 bit. I also tried disabling firewall on Windows 7 but it didn't help.

alperstein, 2015-12-06:
Hi, same problem here. Acat doesn't run on my Win7 64 dell5110 computer

mattengel, 2015-12-06:
I was setting up ACAT on a Windows 7 laptop yesterday and had some other
issues as well.  The gesture app was stuck looping as it went through the
calibration process.  I had previously installed and had it working well on
Windows 8.1 just fine.

I'm updating the Windows 7 laptop to Windows 10 right now.  It's a free
upgrade form Microsoft.  It's nearly complete in the installation now.
I'll report back with my results.  Hopefully this will help you out as
well.

If you want to start the upgrade process, there's an icon that appears in
the system tray that lets you put in a request for the upgrade.  After a
short time, it will let you download the upgrade and start up the process.

alperstein, 2015-12-06:
Thanks mattengel! I have installed other assistive communication software on my computer. I m not sure whether upgrading to Windows 10 would be better or not.

samalkah, 2016-01-20:
Hi ! Good news !
I had  the same problem that a lot of people about Vision which doesn't send trigger to ACAT/Tryout/Talk and I've finally solved it.

It's a problem with the port 27015, that is the port used by Vision and ACAT/Tryout/Talk to communicate. I've read in the troobleshooting documentation that it could come from the Windows firewall  which could block this port sometimes, but even after  having disabled the firewall, nothing changed.

I checked in TCPView if the port was listening and I found out that the port 27015 was used by ACAT and an "AppleDevice..." process. This process comes with I-tunes that I had installed once  for a friend. All you have to do is to uninstall I-tunes (and every Apple application which come with it). Or maybe you can change the port used by I-tunes ... but I didn't try.

Enjoy