intel/acat #20: sensors and  trigger system

rajsurve, 2015-12-09:
where can I get trigger system which are shown in video ?
thnx
raj

tchilaviek, 2015-12-14:
Hello, 

wich video are you talking about? can you get the link please?

rajsurve, 2015-12-14:
Sorry not video, its in pdf
https://01.org/sites/default/files/documentation/acat_user_guide.pdf
Page #14
section 3, Quick  start - Off-the-shelf switches

Let me know if you have any idea where to get those.

tchilaviek, 2015-12-14:
Hello,

i think the triggers or switches are supposed to be something like that:

http://www.cree.fr/fr/catalogue/produits-cree-handicap/domotique-controle-envir/contacteurs-handicap.html

buttons that you are supposed to plug in an USB interface that should send a trigger message such F5 key or F12 (so configured on my computer),
if you try to use acat without the camera recognition system the alternative way to use it will be F5 ou F12 on your keyboard,

so we are supposed to use an USB interface that could detect the action of a button (normally plugged in with an audio mini jack plug,), and configure it in the way i noticed before (sending a F5 or F12 key)

I don't know if such devices exist, but I m sure it will be very easy to do with a simple Arduino, 

if youre interested about that idea I can help you

Regards

A.S

closed 2024-11-19 by csavur