intel/acat #21: User Interface Translation!

tchilaviek, 2015-12-09:
Hello everyone,

i'm working on a French translation of ACAT UI,
i'll would like to translate all the menus, settings panel etc in order to facilitate the use of the program for French people, therapist, etc

my knowledge of code, xml, data bases etc is not very big,
but I managed to start with the translation of the smallest part of the UI: the Buttons!

here are some pictures of that first step:

[image]
[image]

the problem is I found out that in the xml files I could translate only the buttons or change the size of the toggle buttons

for the buttons "OK" "Cancel" "Defaults", we only have ton change the field: label="Someting"

but for the other lines that pertain to the text: "Auto-save scanner's last position", "Hide scanners on idle", "Show Talk window on startup", etc, ther are 8 Lines,
for thoses lines the label is: label="M"

when I change that M by something other, nothing happen,

then I tried to change the fontsize="36"   by   fontsize="24"

I found out the toggle button in front of the line, had changed size

So it does not have anything to do with the TEXT of the UI and my question is

where could I look for, in wich files are the English Texts???

thank you very much

samalkah, 2016-01-21:
Hi,
I didn't check but honestly I think the text of the UI is written in the code so I'm not sure you will able ta change it directly in a xml file. If you want to change the language you should have to download the ACAT developer version, find and change every english text and compile the code, something like that (I'm not a real developer so maybe I'm wrong but I guess it should work that way).
But maybe the changes have to be made in one of these files : 
https://github.com/01org/acat/blob/f2b5946dceada574d34160f74a0d13fcf5c695c7/src/Extensions/Default/UI/Dialogs/GeneralSettingsForm.cs
https://github.com/01org/acat/blob/f2b5946dceada574d34160f74a0d13fcf5c695c7/src/Libraries/ACATCore/PanelManagement/Common/TextController.cs

But I'm really interested to have a french translation. I should check that more deeply

LaurentBerger, 2016-04-18:
@tchilaviek @samalkah Do you success in french translation?

samalkah, 2016-04-19:
@LaurentBerger not at all. Unfortunately I'm not a developer and I hate looking in an application code. The translation of the menus are not the most important for me as long as ACAT have a french dictionnary and french prediction word system and I manage to do that.

Why do you need a full french translation ? Do you know somebody ACAT could help ?

LaurentBerger, 2016-04-19:
I hope i wiil have a french dictionnary in june. Actually there is an italian, spannish and englis dictionnary (in :\Program Files (x86)\presage\share\presage). If you want to try yourself you can follow this wiki https://github.com/01org/acat/wiki/Changing-language-and-creating-new-dictionnaries.

About french translation I can compile and execute code so full translation of menu and dailog box is not a problem but I haven't got enough time

tchilaviek, 2016-04-20:
So we keep in touch ! If you manage to make the UI translation , please let us knoW! Thx

tchilaviek, 2016-04-20:
The aim is to help French users in hospitals or other places dedicated for disabled people for accessibility or creativity , open source interfaces projects in France etc , if We can make the translation it will be easier to work with French people especially if they do not work in computers etc

LaurentBerger, 2016-05-26:
@tchilaviek I think translation will be fibnished in one week. you can send me  a private email if you want. You can use this [page ](http://www.univ-lemans.fr/fr/annuaire.html)to find my email

saiprasadb01, 2016-07-01:
French version of ACAT is now available.  Please check the new <a href="https://github.com/01org/acat/releases">release</a> and give us your feedback.
Thanks
Sai

rajsurve, 2016-07-02:
> French version of ACAT is now available. Please check the new release and give us your feedback.

Can we do  translation in Indian languages?

saiprasadb01, 2016-07-02:
We haven't really explored the option yet.  As far as the keyboard is concerned, it is possible to create a 'alphabet' scanner in the desired language so the user can type in that language.  The problem is with word prediction.  I don't know how we could to next word prediction in Indian languages.  We could turn word prediction off, in which case, the user would have to type every letter.  It would be slow and inefficient, but at least it's a start.  
Which Indian language(s) were you looking at?

rajsurve, 2016-07-03:
I am working  with kid having cerebral palsy who understands Marathi only.

closed 2024-11-19 by csavur