intel/acat #23: ACAT Vision : missing dll msvcp110.dll

samalkah, 2016-01-23:
Hi,
On one of my computer running Windows 8 64bit, Windows tells me the file msvcp.dll is missing when I'm trying to launch Vision. All the other apps (ACAT abc, Talk, Tryout...) work well.

I assume it's a problem with the Framework .NET 4.5 (but maybe I'm wrong) so I tryed to re-install it but nothing changed.

I also tried to download manually the missing dll to put it in the System32 directory but now the program tells me another dll is missing, so I think it will be endless if I keep downloading each missing dlls.

Any suggestion ?

brlima94, 2016-01-24:
@samalkah that should be fixed after you install C++ Redistributables following these steps:
1. [Open this page] (https://www.microsoft.com/en-us/download/details.aspx?id=30679)
2. Click on Download
3. Select VSU_4\vcredist_x86.exe and click Next.
4. Run the downloaded executable.

> Instructions taken from [ACAT Troubleshooting Guide] (https://01.org/sites/default/files/documentation/acat_troubleshooting_guide_2.pdf)

samalkah, 2016-01-24:
@brlima94 thank you but I did that but it didn't work. That's not the same dll, my missing dll is MSVCP110.dll and the troobleshooting guide has a solution for the MSVCP100.dll

But I'm pretty the solution is not far from this.

samalkah, 2016-02-09:
Ok problem solved. I had a lot of C++ Redistributable and .NET Framework versions. I uninstalled eveything then followed these steps : 
following these steps:
1. [Open this page](https://www.microsoft.com/en-us/download/details.aspx?id=30679)
2. Click on Download
3. Select VSU_4\vcredist_x86.exe and click Next.
4. Run the downloaded executable.

Instructions taken from ACAT [Troubleshooting Guide](https://01.org/sites/default/files/documentation/acat_troubleshooting_guide_2.pdf)

saiprasadb01, 2016-08-08:
ACAT [v0.99.1](https://github.com/01org/acat/releases) has been released with new features and bug fixes, including this one.

closed 2024-11-19 by csavur