intel/acat #26: How to connect switch for F12

FrancescoGori, 2016-03-19:
Hi,

how to connect a switch in order to send the F12 key to ACAT?

A friend of mine has this switch:
http://www.auxilia.it/site/lang/it-IT/page/18/product/147

Ideally this USB adapter should work but doesn't allow to send F12:
http://www.bjliveat.com/access/usb-switch-interface-2.html

Any suggestion or alternative?

Thanks in advance

nanomo, 2016-03-19:
Great piece of hardware.

i was reading the documentation and for the USB adapter you can download an extra software to configure the action keys for the switch take a look at the docs:
http://downloads.bjliveat.com/manuals/BJ-805/BJ-805_manual_en.pdf

In the 7_ item of the PDF you will see this:

"**7_ Other applications**
If you want to assign different actions to switches you can do it
with the software Switch Driver from Sensory Software. You
can download it from Sensory Software website:
http://www.sensorysoftware.com/downloads/Swit
chDriver4.zip"

Im making my own switch as are cheaper starting to use old keyboards and mouses in combination with the HotKeys Software its very easy to adapt any middle mouse click to simulate F12 with this you will get a cheaper switch i think :)

For future reference, The HotKeys code will be

```
MButton::
SetKeyDelay,300
Send {Blind}{F12 DownTemp}
Send {Blind}{F12 Up}
```

im doing a tutorial series about all i learned tweaking ACAT and making some custom hardware.

If you need more help dont easitate to ask again

FrancescoGori, 2016-03-21:
Thank your quick reply and your suggestions, the mouse idea is great.

Now I'm building a low-cost bluetooth switch adaptor by modifying a bluetooth mouse.

For converting middle mouse click to F12 I'm using "X-Mouse Button Control" because I've not found the one you suggested. Anyway it seems to be adequate to my purpose.

nanomo, 2016-04-19:
@FrancescoGori 
For the record, AutoHotkeys can be downloaded from:
https://autohotkey.com/

download and install, then start the program, it will stay in the status bar near the clock, then create a file called acat.ahk and pase the code:

```
MButton::
SetKeyDelay,300
Send {Blind}{F12 DownTemp}
Send {Blind}{F12 Up}

With that you are emulating F12 key press when the person click the middle button of the mouse, its the cheapest way to use acat :D, im running a donation campaing in my town to collect mouse and keyboard and make cheaper switchs 
```

tchilaviek, 2016-04-20:
Easy To build with an Arduino Leonardo set as an usb keyboard!

FrancescoGori, 2016-04-24:
@nanomo Yeah, I did it and it's working fine, thanks for the macro.

Also, AutoHotkeys allow to generate a .exe implementing defined macros, very easy to launch and move. Note that on Windows 7 I had to execute it with administrator role in order to send translated key to ACAT.

FrancescoGori, 2016-04-24:
I think that the main advantage in modifying a mouse is that it's trivial to:
- find a mouse, also bluetooth. Limiting new wires may be relevant if the user already has "hardware" on him
- install. Basically any mouse is fully compatible with any OS and doesn't require additional drivers
- modify it for the switch, if you have a minimum of competencies
- explain to someone that doesn't have technical competencies how to use it

MarceloTomaz, 2016-04-25:
Hello everybody!
I'm testing Arduino Pro micro, keyboard simulation is running in windows, but ACAT is not recognized F12.
I changed Actuators ID value with the Arduino GUID, but also without success.

Any idea ?

"Google Translate"  :D

saiprasadb01, 2016-07-01:
New release of ACAT (v0.99) is now available <a href="https://github.com/01org/acat/releases">here</a> with new features and enhancements.
There is a link to the Troubleshooting guide there which has instructions on changing the default trigger key from F12 to another function key.
Thanks
Sai

chicobjr, 2016-11-05:
I could send F12 using Arduino Leonardo, using this circuit https://www.arduino.cc/en/Tutorial/KeyboardMessage
and this code:

```
const int buttonPin = 4;
int previousButtonState = HIGH;

void setup() {
  pinMode(buttonPin, INPUT);  
  pinMode(13, OUTPUT);
  Keyboard.begin();
}

void loop() {
  int buttonState = digitalRead(buttonPin);
  if ((buttonState != previousButtonState)
      && (buttonState == HIGH)) 
  {    
    digitalWrite(13, HIGH);   // turn the LED on 
    Keyboard.press(KEY_F12);
    delay(100);
    Keyboard.releaseAll();
    digitalWrite(13, LOW);    // turn the LED off
  }
  delay(100);
  previousButtonState = buttonState;
}
```
``

alexandre-mbm, 2016-11-09:
By using ATtiny85 USB [something](https://github.com/digistump/DigisparkArduinoIntegration/blob/master/libraries/DigisparkKeyboard/examples/Keyboard/Keyboard.ino) like:
```java
DigiKeyboard.sendKeyStroke(KEY_F12);
```

nanomo, 2017-01-05:
@maeldd 

Install Hotkey software, then create a file with notepad and paste this inside:

MButton::
SetKeyDelay,300
Send {Blind}{F12 DownTemp}
Send {Blind}{F12 Up}

save the file with ahk extension, ex.: keyBinding.ahk and then double click to make HotKey catch and execute the events

nanomo, 2017-01-05:
im making a set of low cost switch, sometime in the test some switch keep turned on, to avoid a false positive i set a delay in the key stroke :), both alternative works mine adds a delay for those reasons

alexandre-mbm, 2017-01-06:
@nanomo, if your set uses a AVR microntroller, you should program a debounce for it. Se http://blog.vidadesilicio.com.br/arduino/basico/botoes/ for more info.

closed 2024-11-19 by csavur