intel/acat #27: acat using rear facing camera

steamer-ned, 2016-03-24:
I have ALS and am trying to prepare for not being able to use a mouse .
I know very little about coding . I have a surface pro 3 iCore7 128 gig with Tobii eye tracking.
I find the eye tracking unreliable and erratic so i install acat.. when I start acat vision it picks the rear camera . I have read instructions and searched for a way to switch but no luck. please direct me to where I can find an answer

LaurentBerger, 2016-04-18:
I have a surface pro 4 and I have got same problem. If I plug a logitech webcam (c270 a basic) ACAT use this webcam and rear or front camera.

nanomo, 2016-04-25:
I will try to help both but i need to see your setup to understand the underlaying problem, if you want add me to skype: mariano.montanez , in the message let me know you wanthelp for ACAT

steamer-ned, 2016-04-26:
I don't understand why you need setup,.I have ALS s I can't take pictures
of the setup. I have a surf pro 3. I have it in mount, sitting on a
hospital table and use a mouse to run the computer
I run the acat vision  and It selects the rear facing cam instead of the
front camera so I see the instead of my face. There are no provision for
selecting switching cameras.
I have switched the cameras before running the software but it still
selects the rear facing cam. I don't code very well, but there be something
in the software that makes the rear camera default.
Please help, thanks John

brlima94, 2016-04-26:
Hi @steamer-ned and @LaurentBerger,

Could you ask for someone to open Windows' Device Manager, go to Imaging Devices, disable the rear camera and restart your computer?

This should make the front camera be the only one available for ACAT.

saiprasadb01, 2016-07-01:
New release of ACAT v0.99 is out with new features and enhancements.  You can download it from <a href="https://github.com/01org/acat/releases">here</a>.
If you have multiple cameras, ACAT will now let you select the preferred camera.
Thanks
Sai

saiprasadb01, 2016-08-10:
Closing issue.  The current version of ACAT lets you select the preferred camera

closed 2016-08-10 by saiprasadb01