intel/acat #28: Running ACAT as a user

LaurentBerger, 2016-04-19:
Hi,

Is it possible to run ACAT as a user?
I have installed ACAT as administrator. I can run acat but I cannot run as a simple user.

Thanks for your answer.

nanomo, 2016-04-19:
did you receive any message?

LaurentBerger, 2016-04-20:
No I didn't

saiprasadb01, 2016-08-08:
ACAT doesn't really need admin rights to run.  To which folder did you install ACAT?  Did you install it to the default location c:\intel\acat?   And what happens when you run as a user?

LaurentBerger, 2016-08-09:
I have done a standard install but when I run acat it cannot found presage database (in an user account). Using admin account on same machine acat run. (Windows 10 pro)

saiprasadb01, 2016-08-10:
Maybe you can go into Windows User Manager and add the user to the Administrator group. Go to Control Panel, select "User Accounts"

closed 2021-03-02 by LaurentBerger