intel/acat #29: LOCALIZATION

brunopelegrini, 2016-05-31:
Hello , we managed to finish the translation of the ACAT system for Brazilian Portuguese , we will make it available at the following link, http://brlima.com.br/acat if someone is interested !

[image]

Thanks for your help .

brunopelegrini, 2016-05-31:
[image]
[https://www.youtube.com/watch?v=pJo78ZX_Qk4(url)

rajsurve, 2016-05-31:
Can this be done in any other language like Hindi ?

TomazAlexandre, 2016-06-08:
Enquanto não sai essa versão final ai você pode traduzir o dicionário de palavras para português.

https://github.com/01org/acat/wiki/Criando-um-novo-dicion%C3%A1rio-Pt-Br-e-Alterando-o-idioma-ACAT

brunopelegrini, 2016-06-08:
Tomaz, já efetuamos a tradução do dicionário para diversos idiomas, dentre eles estão, português (pt-BR), Francês, Italiano. Espanhol e Polonês, nosso projeto foi desenvolvido perante a necessidade dos usuários a utilizar o sistema com sua interface gráfica traduzida, principalmente no primeiro momento nas configurações e na síntese de voz, como dito anteriormente o sistema está 100% traduzido, porém estamos aguardando retorno da Intel para a liberação do mesmo.

Agradeço pelo apoio.

saiprasadb01, 2016-07-01:
Really excited about the Portuguese version.  Well done!  How did you do the word prediction?  Did you create a Presage database for the Portuguese version? 
We released v0.99 yesterday with Localization support.  If you can send us the Portuguese language resource file and the word prediction database, we will create a Portuguese language pack and upload it on the ACAT release page.
Thanks
Sai

gabrielarabelo, 2016-07-07:
Hi all,
My partner and I are also interested on contributing on the development of a PT-BR language module.
I work on Algorithm Programming for AI and my partner works with Corpus Linguistics.
There are some open and free corpora of both spoken and written Brazilian Portuguese and we may in the next few weeks try to mine and organise them.
I still couldn't check in details how to implement the WP/Presage it but as I noticed everything is very well documented and we may not find it hard (I hope) - if someone has already started any effort on that, please let me know.

Sai, as I understood, Bruno has translated the user interface commands but not the WP yet.

Bruno, vocês já estão trabalhando na questão da word prediction ou ainda não começaram? Seu grupo estaria interessado em juntar esforços?

Cheers for now,
Gabi

brlima94, 2016-07-07:
Hi @gabrielarabelo ,

First of all, thanks for your interest.
Both UI and word prediction in Portuguese were finished.
We also created custom keyboards based on ABNT2 Layout.

Now we need to adjust everything to work as a language pack when the latest version's source code is released.

Would you like to build a speaking engine or customize any existing one to work with ACAT in Portuguese?
Maybe it's only me, but I find the Windows 's Portuguese accent a little weird.

brlima94, 2016-07-07:
Hi @saiprasadb01 ,

We created a presage database, but we needed to change some code in order to support diacritics.

We'll send you the translated resx file, but what can we do about the diacritics and custom keyboards?

Will you accept a pull request or we need to email you the modified files?

gabrielarabelo, 2016-07-07:
hey @brlima94, that's great news!
Did you use any corpus to create the database or you translated the english one?
The diacritics support was certainly an essential step to get into. I am glad to hear you already came across this part.

saiprasadb01, 2016-07-08:
Hi @brlima94 
If you install the French Language pack, you will see how diacritics and custom Alphabet keyboards are handled by ACAT.  After installing the French language pack, go to the C:\Intel\ACAT\fr folder.  Under that you will find LetterA.xml, LetterE.xml, LetterC.xml etc. All these scanners have letters with diacritics for the letter A, E and C.  Next look at PanelConfigMap.xml in the same folder.  There are entries in there with config entries for these scanners.  To see how the scanners with diacritics are displayed, look at AlphabetScannerAzerty.xml and search for showPopup().  For instance, for Button B3, for the letter A in French, it shows the LetterA scanner which contains all the diacritics for the letterA.    If you scroll down that file, Look for <Animation name="ButtonRotationRowL1" >  Under that, for button B3, there is a call to showPopup().
This is how ACAT deals with diacritics.  Use an XML editor or Notepad++ to make it easier to edit the files.
Thanks
Sai

saiprasadb01, 2016-07-08:
Hi @brlima94 ,
I have the same question as @gabrielarabelo .   What did you use as the source material for the Presage  database?  Also, can you point us to where we can downlaod the database from?
Thanks
Sai

brlima94, 2016-08-05:
Hi @saiprasadb01  and @gabrielarabelo ,

The database was created from a random text in portuguese (can't remember the source right now).
I created a tool to automate the conversion from **.txt** to **.db** (source code available [here](https://github.com/brlima94/presage-dictionary-builder)).

If you're interested in the database files, you can get them from [this branch](https://github.com/brlima94/acat-localization), source code is now updated.
There is also a binary version available [here](https://github.com/brlima94/acat-localization/releases/tag/v0.98).
Please note this was made based on ACAT 0.98.

This is the same app presented in the video posted by @brunopelegrini 

@saiprasadb01 , I created a Slack room for us to talk about the Portuguese Language Pack and User Guides for version 0.99.
@gabrielarabelo Would you like to join us? Send me an e-mail (same login, at outlook . com)

saiprasadb01, 2016-08-08:
I would love to join the slack room to discuss Portuguese Language pack.  Let me know how.
BTW, we released [v0.99.1](https://github.com/01org/acat/releases) with new features including a Spanish language pack.  The release page has the English resx file with pointers on how to translate..  If you can translate the strings to Portuguese, I can use your existing Presage database file and release the Portuguese language pack.

saiprasadb01, 2016-08-08:
Hi @gabrielarabelo, @brunopelegrini,  Sorry I missed the Slack invite you had sent me.  I just joined.  Thanks

zsmarton, 2016-08-24:
@saiprasad01 I have just sent a Hungarian database and the translated ACATResources.en.resx file to your mailbox, and I kindly ask for a Hungarian language pack for ACAT. It would mean a new life for my 12 year old step daughter, who was born with tetraparesis.
Thanks a lot.

saiprasadb01, 2016-08-24:
Wow!! Thanks so much.  We will release the Hungarian language pack for ACAT.
BTW,  is the Hungarian keyboard is a little different from the English QWERTY keyboard.
Is this the correct layout?  The letter Z is interchanged with the letter Y

[http://ascii-table.com/img/keyboard-208.png]

saiprasadb01, 2016-08-24:
What text did you use as the source to create the database?

zsmarton, 2016-08-25:
Hello,

Thank you so much.

Yes, this is the Hungarian keyboard. Z and Y interchanged.

The sources of the database are:

Egri csillagok, a 19th century novel set in the era of the  Osman
occupation of Hungary in the 16th century. It has been required reading in
primary school for decades, and is pretty much part of the Common Hungarian
knowledge. Around 600 pages, a good basic source of words, I thought.

Another source was a Maths book for 12 year old pupils, for a different
kind of language. And next I choose a contemporary novel for teenagers: The
Princess Diaries by Meg Cabbot, for the contemporary dictionary of everyday
life.

Of course, there could be better sources for general usage. I kept this
little 12 year old in mind when I was peraring the database file. Probably
I could ask some linguist friends of mine for suggestion of sources for a
good general database.

Thanks again,

Zsuzsa

saiprasadb01, 2016-08-25:
Can you take a look at this source and see if can be used to create the database?
http://mokk.bme.hu/resources/hunglishcorpus/

zsmarton, 2016-08-30:
**@saiprasadb01** An email is sent to you with a large, legal database in Hungarian. Regards, Zsuzsa

molerat619, 2016-11-12:
How did you guys do that? I want this in German for my cousin. I can't find the right presage database.

molerat619, 2016-12-17:
Has anybody started working on a German database yet? If not, I would start doing so. 

@brlima94  Could you invite me to Slack too for support? marcel.izgin@live.de

Would be amazing if I can get this done for my cousin by christmas :)

brlima94, 2016-12-21:
Hi @molerat619,

Invite sent as requested. Please check the database file I posted at #7 based on the text file you provided.

Good luck with your Christmas gift.

closed 2024-11-19 by csavur