intel/acat #30: UI y Diccionario en Español

EmiLafu, 2016-06-15:
Buenas tardes para toda la comunidad, mi nombre es Emiliano Martin Lafuente, no se si realmente ya se realizo, pero acabo de terminar de traducir toda la UI y configurar para que todos este en español el ACAT. La fonética la utiliza del narrador de windows así que queda en cada uno cambiarla, los usuarios de Windows 7 no la van a poder cambiar por que solo trae una y es la de Anna, pero tranquilos que se pueden bajar la de Jorge (Español  - España) e instalarla. Ahora les muestro un par de capturas de pantallas para que vean mas o menos como queda.
[image]
[image]
[image]
[image]

Bueno les dejo los links para descargarselo:
acat.zip:
https://drive.google.com/open?id=0B_EUOrlEnWqSVHR5NXRlNFhJRDA
Presage.zip:
https://drive.google.com/open?id=0B_EUOrlEnWqSbnRRUmJyZFp1Szg

La instalación se hace de la siguiente manera:
Descargar e instale el programa Acat oficial, una vez finalizada la instalación, descomprimir los archivos zip modificados, copiar y reemplazar las carpetas en sus correspondientes rutas de instalación.

El presage se instala en C:\Program Files (x86)\presage *1
El ACAT se instalo en c:\INtel\Acat

*1 - cuando instalan el Acat oficial se instala tambien el Presage, este se activa ni bien se termina la instalación, asi que al momento de reemplazar la carpeta cierrenlo asi no les apareceran errores.

Espero que les sea útil, cualquier inconveniente o modificación avisen.

Saludos a todos.
Emiliano Martin Lafuente.

brlima94, 2016-06-16:
Hi @EmiLafu,

I am working on a multilingual version of ACAT.

I already have UI and word prediction fully working in Portuguese and word prediction (without a custom keyboard and menus) in Spanish, French, Polish and Italian.

Can I extract the files you translated and use it to display the UI in Spanish?

Also, would you like to help me define the keyboard layout for ABC and QWERTY?
If so, please take a look at http://brlima.com.br/acat/andamento.html for examples.

There are also some hardcoded strings in the source files, but we can talk about that later (I'll put all of them inside a worksheet and send you, but it will take some time).

EmiLafu, 2016-06-16:
Hi @brlima94,

Yes, you can use the files translated for your work, only asks you to make mention of my work.
I will do my best to help you define the layout of the keyboard querty and ABC.

NIce work and good luck.

brlima94, 2016-06-16:
Hi @EmiLafu ,

Sure, I'll mention your work on the splash screen and also inside ACAT's About screen.
I can also include a link to any email or website, if you want.

Could you please take a look at that link and see if we can use the same keyboard layout for Spanish, changing just the last row?

EmiLafu, 2016-06-16:
Hi @brlima94,

Yes I saw the images of the link , I'll see if I can make modifications on keyboards.
What I realized is that I not translated the ACAT vision, now I'm working on it.

EmiLafu, 2016-06-16:
Hi @brlima94,
can you tell me what size you assign to the keys on the keyboard querty?, to make it equal to your design.

brlima94, 2016-06-16:
@EmiLafu it's Arial 22

nanomo, 2016-06-16:
estamos todos en la misma, no tuve tiempo aun de compartir pero llegue a lo mismo que @brlima94, ya que este repo no recibe pulls porque no seguimos en el de @EmiLafu ?

yo estuve trabajando durante los ultimos meses con personas con distintos problemas, les comento que una de las necesidades con las que se van a topar es poder agrandar la letra de toda la UI al menos de la escritura, como ven poder hacer ese cambio?

EmiLafu, 2016-06-16:
Hola @nanomo,
Una salida rápida y poco ortodoxa, jeje, de solucionar el problema de las personas que necesiten incrementar un poco la letra, sería cambiandole la resolucion a la pantalla, el tema de agrandar las letras seria modificar todas las dimensiones de la UI. La personalización del teclado al español como lo hizo @brlima94 lo estoy realizando.

brlima94, 2016-06-18:
Hi @nanomo,

You can change the whole UI size by going into ACAT's Main Menu -> Settings -> Size/Position.

brlima94, 2016-06-23:
Hi @EmiLafu  and @nanomo ,

I made a preview build with spanish. You can check it out [here] (https://drive.google.com/open?id=0B0HCqIsdF4ribXgwMlJJaDA4LTQ) (you may need 7-zip in order to open the file).

The menus are partially translated (thanks to @EmiLafu ).
I still didn't add the credits inside the app, but I'll will do so probably on next build.

The database was created from a random text file found on internet, if you have a better sample text please let me know.

The keyboard was based on the Brazilian Portuguese keyboard, I just replaced "**ç**" by "**ñ**".
Is it OK if I just remove the characters "**ã, à, â, ê, ô**" from the keyboard, or there's something else I need to change in it?

nanomo, 2016-06-23:
> Is it OK if I just remove the characters "ã, à, â, ê, ô" from the keyboard

@brlima94   There are important chars, think in the people using ACAT to write important text, they will need all the chars :)

brlima94, 2016-06-30:
@nanomo OK, so I won't remove them.

@EmiLafu and @nanomo  ,
Were you able to test the app using the link I posted?

saiprasadb01, 2016-07-01:
ACAT v0.99 released yesterday with localization support.  We will release the source for this version in a couple of weeks.  All the ACAT strings are in now in resource files.  We will uploaded the English resource .RESX file to the ACAT release pate.  If you can translate the strings in there to Spanish or any other language and send us the translated .RESX File, we can generate a Language pack and make it available on our <a href="https://github.com/01org/acat/releases">release</a> page.   Please check the instructions on the  <a href="https://github.com/01org/acat/releases">release page</a> for helping us with localization.

Thanks
Sai

nanomo, 2016-07-01:
Im translating the resx, please for future changes use git there are a lot of effort wasted :(

nanomo, 2016-07-01:
@EmiLafu want to jon in resx translation? https://github.com/nanomo/acat-es-resx

EmiLafu, 2016-07-01:
@nanomo Hola Mariano, estoy traduciendo el .resx con SImple Resx editor hace rato ya, pero no quiero que tu esfuerzo sea en vano tal como lo mencionas, pero tampoco veo como unirme a tu traducción, que propones?

nanomo, 2016-07-01:
si llevas mas sobreescribilo si terminas bien sino lo sigo yo, avisame :D

EmiLafu, 2016-07-02:
just I finished translating all strings to Spanish, hope it can be useful for language pack.

EmiLafu, 2016-07-02:
.resx in Spanish
https://drive.google.com/open?id=0B_EUOrlEnWqSNW9vWndxZEhqWG8

saiprasadb01, 2016-08-08:
New pre-release of ACAT [v0.99.1](https://github.com/01org/acat/releases) has been released, includes Spanish Language pack and other new features and bug fixes

EmiLafu, 2016-08-09:
Thank you Sai for this update and thanks to all the team , Greetings from Argentina

closed 2016-08-09 by EmiLafu

alexandre-mbm, 2016-08-09:
Source code please. Hint: new branch, temporarily.