intel/acat #31: New pre-release of ACAT (v0.99) now available

saiprasadb01, 2016-06-30:
Localization support added, plus a host of other features.  Check out the release notes on the  <a href="https://github.com/01org/acat/releases">release page</a>.  Documentation has also been <a href="https://01.org/acat/documentation-list">updated</a>.
This release includes the base English version and a French version which can be downloaded separately as a Language Pack.
This is a binary-only pre-release.  The source code and the official release will be in a couple of weeks.
The User Guide for Localization will be uploaded shortly which will cover details on translating strings, creating scanners etc.
Please let us know of issues/problems so we can address them before the final release.

saiprasadb01, 2016-07-01:
Updated the <a href="https://github.com/01org/acat/releases">release</a> page with instructions for you all to help us with translating ACAT to other languages.  
We welcome all the help we can get to translate ACAT into other languages.

LaurentBerger, 2016-07-02:
@saiprasadb01 Thanks for this new release and this project which is really nice and usefull.  

Many people have started to work on this project to improve it  language translation, sensor... It would be really nice if everybody can make pull request on this project to share work with communauty

Thanks again for this work

closed 2024-11-19 by csavur