intel/acat #33: Why the software history was not kept?

gustavopinto, 2016-07-15:
Hi there,

I'm a researcher studying software evolution. As part of my current research, I'm studying the implications of open-sourcing a proprietary software, for instance, if the project succeed in attracting newcomers. However, I observed that some projects, like _acat_, deleted their software history.

https://github.com/01org/acat/commit/39577a1248677d1fb8b64af9d517fb371756688e

Knowing that software history is indispensable for developers (e.g., developers need to refer to history several times a day), I would like to ask acat developers the following four brief questions:

1. Why did you decide to not keep the software history?
2. Do the core developers faced any kind of problems, when trying to refer to the old history? If so, how did they solve these problems?
3. Do the newcomers faced any kind of problems, when trying to refer to the old history? If so, how did they solve these problems?
4. How does the lack of history impacted on software evolution? Does it placed any burden in understanding and evolving the software?

Thanks in advance for your collaboration,

Gustavo Pinto, PhD
http://www.gustavopinto.org

gustavopinto, 2016-08-22:
Hi @saiprasadb01, do you have any comments?

markilling, 2016-08-23:
Hello Gustavo,

 

I am currently traveling and my internet is irregular.  Apologies if my interactions are slow.

 

Are you currently the one supporting the ACAT software?  I am looking at efficient means of text entry for people with severe mobility impairments.  I just learned that the ACAT software, developed for Stephen Hawking, is open source, so I am looking at it as time allows.  I am intrigued at what you have, but I am a little surprised.  My background is in developing T9 and Swype for mobile phones.  I am curious how the ACAT software differs from Stephen Hawking’s.  I would have expected him to be using a disambiguation engine (I thought he was using SwiftKey’s) with the advantage of fewer keys to scan through.

 

In any event, I am considering whether or not to adapt the ACAT software to include ideas we have written about.  We would want to make significant extensions for input devices and use of a disambiguation engine.  Besides scanning, we see support for keypads, joysticks, guarded touchscreens and eye gaze systems.  I am speaking with Matteo Vescovi about what would be required of Presage to support ambiguous keys.

 

The advantage of doing this is that the scanning space is so much smaller.  If you are interested, you can see the paper we presented at CSUN this past spring here:  <http://www.continuouspath.org/app/download/763979208/CSUN2016JustType.pdf> http://www.continuouspath.org/app/download/763979208/CSUN2016JustType.pdf.  It describes efficient text entry and computer control using 8 keys where text can be entered using 0.7 – 1.02 keys per character – the variation depends on the amount of cognitive load the user chooses.

 

Anyway, I do have many questions about ACAT and would love to discuss them if you are willing

 

Mark Illing

Product Development

Continuous Path

2136 N 62nd St.

Seattle, WA  98103   USA

+1 206 799-9572

 <http://www.continuouspath.org> www.continuouspath.org

 

“Pursue something so important that even if you fail, the world is better off with you having tried.” — Tim O’Reilly

closed 2024-11-19 by csavur