intel/acat #39: Mouse Scanner issue and ACAT losing focus

lonewolvf, 2016-08-08:
I am unable to select anything along the edges of the screen due to my inability to click fast enough before the rectangle goes in the opposite direction or the line restarts scanning.  Could you make it so the rectangles and lines stop for a brief amount of time at the edges?  

There are times when ACAT becomes unresponsive.  I think it loses focus of current application when that happens.  If that is the issue, could you have ACAT focus on the desktop as a backup default?

Thank you,
Michael Huehmer

alexandre-mbm, 2016-08-09:
@lonewolvf, are okay your triggers? Do you use USB switches, Bluetooth or Vision?

lonewolvf, 2016-08-09:
I use a USB switch box.  StealthSwitch AT 10 ports http://www.stealthswitchat.com/

closed 2024-11-19 by csavur