intel/acat #40: New pre-release of ACAT v1.00.0 now available

saiprasadb01, 2016-10-31:
New features added, bugs fixed from the previous release.  ACAT for Portuguese is also included.  Click [here](https://github.com/01org/acat/releases)

chicobjr, 2016-11-02:
Hi Sai Prasad,

I have installed version 1.00.0 on a PC with WIndows 10 and selected Portuguese language as default. This version seems to be more instable than previous one. Here are some of the problems I'm facing.

1) ACAT Tryout is crashing when exit, here is the stacktrace:
System.NullReferenceException: Referência de objeto não definida para uma instância de um objeto.
   em ACAT.Extensions.Default.UI.Dialogs.YesNoDialogForm.Form_Closing(Object sender, FormClosingEventArgs e)
   em System.Windows.Forms.Form.OnFormClosing(FormClosingEventArgs e)
   em ACAT.Extensions.Default.UI.Dialogs.YesNoDialogForm.OnFormClosing(FormClosingEventArgs e)
   em System.Windows.Forms.Form.CheckCloseDialog(Boolean closingOnly)
   
2) ACAT Tryout is selecting the first letter as soon as it is opened, even if I remain still (I'm using visual actuator). In addition, it is not rare that it selects the first letter more than once, as if I was triggering repeatedly. It seems to me that facial movement detection is very sensitive, I am not sure if there is any setting I could change for that.

3) Talk applications are crashing when exiting, but it is not possible to get the details (stack trace); when this happens, every other app on Windows stays on background, so that it is not possible, for instance, to open a Task Manager and kill processes - the only way out I could find was to logoff and login again.

I am familiar with C#, I have VS 2015 installed on this PC; if you put the latest source code, I think I could be able to give you more information (the latest source code is for version 0.98)

Thanks.

chicobjr, 2016-11-07:
More information about my comments above: problems reported do not happen if I select Keyboard Actuator instead of Vision.
Another issue: opening Talk (Alt) and Talk (Alt ABC) and selecting one of the suggested words on the bottom-left panel inserts a different word on the Talk Window.

zelmon64, 2017-04-24:
> Another issue: opening Talk (Alt) and Talk (Alt ABC) and selecting one of the suggested words on the bottom-left panel inserts a different word on the Talk Window.

@chicobjr did you manage to figure this out? I found the error is in the .xml files. For example [here in the AlphabetScannerAlternate.xml file](https://github.com/01org/acat/blob/master/src/ACATResources/en/AlphabetScannerAlternate.xml#L228-#L235).

The correct form is:

`<Widget name="PWLItem1" onSelect="actuate(PWLItem1);transition(TopLevelRotation)"/>`

closed 2024-11-19 by csavur