intel/acat #43: Cannot select voice in V1.0

molerat619, 2016-12-21:
I am running ACAT App on Windows 10 in Version 1.0.0 and I cannot change the voice for TTS.

In my windows settings, I have following voices available (german):

[image]

Stefan is selected and in my ACAT configuration I picked **male**, but I can't select a voice in the top:

[image]

What can I do here? Thanks.

zelmon64, 2017-04-23:
@molerat619 Have you had any luck with this? I found it can be changed by selecting the computer's default voice in the Control Panel. Search for  "text to speech settings" (or the german equivalent) to find it:

[image]

sofiamf, 2017-09-03:
Did you find a solution for this? @molerat619 @zelmon64 
I was able to add the portuguese language pack to the options
[image]
However, ACAT is ignoring this settings and after use I go check this windows and it has been changed.

Thanks in advance.

molerat619, 2017-09-04:
@sofiamf @zelmon64 Sorry, but no, I have not worked on this for a while and the latest state was that it did not work for me. Maybe now with the latest version of ACAT it works. I will let you know once I try it again.

sofiamf, 2017-09-04:
@molerat619 Okay, many thanks! Let me know then please :)

RafaelScience, 2018-05-20:
I could not find the solution to solve this problem, could anyone?

closed 2024-11-19 by csavur