intel/acat #44: Blink detection actuator switch

unixunion, 2016-12-22:
I have a family friend who could use ACAT if it can do blink detection as a "switch". Is there any possibility to add "blink" as a actuator switch.

alexandre-mbm, 2016-12-24:
You can use any switch (hardware or software) by binding it like a simulated keypressing for F12 key.

RalfCT, 2017-04-18:
Did you manage to find a 'blink' actuator and configure it as an F12 trigger?  I have an MS friend who may also have to rely on blinking as most of her other faculties are lost.

unixunion, 2017-04-19:
No I didn't find any sw that could do the blink detection. :/ I reached out
to many people but seems like it's not something anyone wants to build.

/K

RalfCT, 2017-04-19:
Hi Kegan,

I've also downloaded Click2Speak as an alternative to ACAT.  Check out 
the following, eye tracking may be an option for you.

  * http://www.click2speak.net/category/our-journey
  * https://tobiigaming.com/products/

Cheers,
Ralf

ahaleksandra, 2017-06-21:
Hi,
I have tried Emotiv Epoc+ but I cannot make it works with ACAT... Is there anyone who could give me any advice how to connect this...? 
I am looking for solution for woman who has MS... We think that blinking should be the best... I will be glad for any hint how to do it... 
Thank you indeed!

zelmon64, 2017-06-23:
@ahaleksandra I've made a blink actuator using OpenFace. It sends F22 so that it shouldn't conflict with any thing so add that as another trigger to the keyboard actuator.

[OpenFace_Blink_Actuator_v0.0.1_x86](https://drive.google.com/open?id=0B7SYYOqNW4FSWDJXWlNWR3BSWUk)
[OpenFace_Blink_Actuator_v0.0.1_x64](https://drive.google.com/open?id=0B7SYYOqNW4FSanpOXzc3MGZXb3M)

You'll probably need to install the [Visual C++ redistributable package](https://www.microsoft.com/en-us/download/details.aspx?id=48145).

- q to quit
- r to restart
- s to display the current aspect ratio of the left eye
- d to display the current aspect ratio of the right eye

the threshold ratio is currently set to 4.

LaurentBerger, 2017-06-24:
@zelmon64 thanks for code sharing. 
I have read your code and to send a F22 do you use only boost only ?

zelmon64, 2017-06-24:
@LaurentBerger it uses [SendInput](https://github.com/zelmon64/OpenFace/blob/Blink_Actuator/exe/FaceLandmarkVid/FaceLandmarkVid.cpp#L214). I'm not sure what you mean about using boost. I only copied an example I found. If there's a better way I'll happily change it.

closed 2024-11-19 by csavur