intel/acat #45: No Support to PT-BR (Limitation)

RafaelRMachado, 2017-01-20:
HI guys
We already created the needed files but didn't receive any feedback by e-mail

So we're opening this as a limitation.
Please try to help us to support pt-br because we have several people in Brazil that can use this tool.

Thanks
Rafael
[ACATResources.pt_BR.RESX.zip](https://github.com/01org/acat/files/719912/ACATResources.pt_BR.RESX.zip)

RafaelRMachado, 2017-01-30:
Hi guys.
Please help us to solve this issue.
We have a lot of people in Brazil that could use this application.

We'll keep waiting

brlima94, 2017-04-15:
Hi @RafaelRMachado, have you tried using this Language Pack (https://github.com/01org/acat/releases/download/v1.00.0/ACATPack-Portuguese.exe)?

RafaelRMachado, 2017-04-19:
Hi Bruno

I had not found this pack..
I'll try it.

Thanks a lot
Rafael

closed 2024-11-19 by csavur