intel/acat #50: Tobii Eye-Tracker and ACAT

RalfCT, 2017-04-26:
Can one integrate a Tobii or The Eye Tribe eye-tracker into ACAT, if yes, how?  Our user can no longer mover her fingers sufficiently to use them as a trigger, nor is a webcam an option due to limited facial movement.

Many thanks in advance.

zelmon64, 2017-04-26:
@RalfCT it can probably be done. What precisely would be used for the trigger? Maybe looking at a designated spot or off screen direction or wink/blink detection? I ask because for less precise facial cues [OpenFace](https://github.com/TadasBaltrusaitis/OpenFace) works reasonably well and uses a webcam if budget is an issue. I once tried to use OpenFace as a gaze mouse which wasn't quite precise enough but as a trigger it should be okay. If you'd like to try it I can make a build of OpenFace that acts as the trigger?

Alternately, if the eye-tracker is used as a gaze mouse it should work with ACAT without any tweaking.

RalfCT, 2017-04-26:
@zelmon64 I'm currently also playing around with Click2Speak (sadly Gal Sont the developer contracted ALS and has since passed away,  Towards the end he did most of the coding using only a Tobii eye-tracker and his software).  ACAT and Click2Speak each have their own unique attributes.  His software uses a 'Dwell' function which 'clicks' the key if one looks at it for a predetermined time using an eye-tracker.  While his software is active one can basically operate all programs on the PC, the TV, A/C etc. using the 'Dwell' method.  It would be great if this functionality could be integrated into ACAT, especially for those who cannot move their limbs or trigger via a webcam.

zelmon64, 2017-04-26:
@RalfCT if Click2Speak let's you use the eye-tracker as a gaze mouse with dwell clicking can't that be used directly with ACAT to type? What would be the benefit of integrating it into ACAT?

Have you also tried [OptiKey](https://github.com/OptiKey/OptiKey)? It seems fairly similar to Click2Speak  but without the TV, A/C etc functions.

RalfCT, 2017-04-26:
@zelmon64 I'm not sure if ACAT will recognize a Tobii (or other) eye-tracker as a gaze mouse once connected and configured.  If it does, then yes, one should also be able to type, much fast than waiting for the software to scan through each block/letter.

zelmon64, 2017-04-26:
@RalfCT ACAT doesn't have a problem with my mouse emulation with a joystick so it should be fine with a gaze mouse. I can't think of a reason why it should behave differently (but I suppose you never really know with these things).

RalfCT, 2017-04-26:
@zelmon64 thanks for the prompt response.  Let me also check out OptiKey, the TV and A/C functions are not critical, after all if the patient is communicating with someone then there will be someone present to assist with the TV etc.  It's sad that ACAT, Click2Speak etc. are not really gaining much traction considering the need, yet other useless videos on YouTube get millions of views

RalfCT, 2017-04-26:
@zelmon64 Just tried out OptiKey, it's very cumbersome, slow and extremely limited when compared to Click2Speak and ACAT.  I also don't like how it re-arranges my icons to accommodate the on-screen keyboard, leaving them squashed at the top of the screen when one quits the program.

zelmon64, 2017-04-26:
@RalfCT you're very welcome. I know what you mean about the lack of activity. I've been trying to find somewhere to discus all these programs but haven't so far - not even on Reddit. Do you know anywhere to discus accessibility software?

Sorry about OptiKey squashing your icons. I think it can be used floating instead of docked but that's probably moot since you find it too cumbersome.

RalfCT, 2017-04-26:
@zelmon64 Re getting the software out there, I posted my positive experience with ACAT on several MS FB forums after joining my wife to visit her friend with MS last weekend (I only got a few responses).  I hadn't seen her since 2011 when we all went for a pizza, now she's bed-ridden, frail and no longer able to communicate.  I'm looking into setting her up with a solution, probably Click2Speak due to her physical limitations.  One could possibly offer the communication solution (hardware/software/training) on a rental basis to other families with MS, ALS etc. patients.  The Intel NUC would be a suitable platform - small, quiet and powerful.  

These are devastating diseases, sometimes I think it's better to have the brain fail while the body remains intact than the body failing while the brain remains intact.

RalfCT, 2017-04-26:
@zelmon64 Having spent some time with OptiKey I must say, it has some interesting features, particularly the multi-key selection option (although it doesn't seem to like starting a sentence with "I" while in that mode).  One basically has three good (and free) Open-Source communication software options for the handicapped, there's no longer a need for them to be without a voice.  

Quite often the caregivers, family members or doctors have no idea what options are available, or where to even start looking.  I've been in IT since 1999, thanks to you I found out about OptiKey, and thanks to a post on the ACAT site I became aware of Click2Speak.  One should have a central site where all the options are presented, together with their pros and cons.

zelmon64, 2017-04-26:
@RalfCT I'm glad OptiKey is growing on you and that Julius was able to solve your problem. I agree a central site would be ideal. With all these Open-Source software solutions it seems a shame the hardware is still relatively expensive. Wouldn't it be nice if it could all be run off something like a Raspberry Pi with a budget webcam?

ahaleksandra, 2017-06-21:
Hi, 
I am glad that you are talking about this here. I need to find some solution for communicate using just facial movement. I would like to try also eye-tracking, so I am looking for sites where people share their ecperience about using differen switches and software solutions. I will be happy also to share my ecperience. If you know this kind of site or want to create one, please let me know - together we can change sth. 
Thanks,

zelmon64, 2017-07-04:
@RalfCT have you seen the latest version of click2speak (1.7.89.0)? It's really fancy!

@ahaleksandra I haven't found anywhere people discuss these software solutions in general. It would be useful.

Juno808, 2018-08-16:
I second this. It would be great if some of eye tracking methods (such as dwelling) could be integrated into ACAT. Intel already has [RealSense Depth Camera]((https://eu.mouser.com/new/Intel/intel-realsense-camera-400/) ) series with infrared sensors that could work as Eye Tracker.

closed 2024-11-19 by csavur