intel/acat #51: Lack of buttons icon

gersonfs, 2017-06-21:
I have installed the latest version of ACAT with the PT-BR translation installed on a notebook with updated Windows 10 64bit. In the first 2 runs all button icons popped up. From the 3rd execution there is the button but there is no icon to identify it as shown below. These same buttons have no icon on all screens. What can this be?

https://ibb.co/mmTusk

gersonfs, 2017-06-27:
Hello @zelmon64

I saw that you are an active contributor to the project, could you help me debug this problem?

nanomo, 2017-06-27:
hi @gersonfs please try the next:

1) change resolution of the screen

2) change the language of the pc

seems a character problem maybe related with encoding but maybe that items to discard alternatives.

zelmon64, 2017-06-27:
@gersonfs well this is strange. I installed ACATPack-Portuguese.exe (that is the PT-BR one right?) and it messed up the icon symbols for both English and Portuguese. I then uninstalled ACATPack-Portuguese.exe l but the icons didn't return so I also uninstalled the main ACAT program (choosing not to delete the files). I then reinstalled both the main ACAT program and ACATPack-Portuguese.exe to recheck (and this is strange part) but now all the icon symbols appear correctly!

In short, I suggest you uninstall all of the ACAT bits (without deleting user data) and then reinstalled them.

gersonfs, 2017-06-27:
Thank you all for the answers. I'll do the tests later in the day. I also installed the PT-BR translation package.
I forgot to mention that I edited the files according to this pull request: https://github.com/zelmon64/acat/commit/391aa8737d6e68ae21633fde8ea9aa7561ab2455

What is the coding of these files? Utf-8, latin1 or other?

zelmon64, 2017-06-27:
I'm also using those modified xml files (obviously) so they shouldn't be effecting anything. They are Utf-8.

Thought I'd add a screen shot just to confirm that it's working for me:
[image]

gersonfs, 2017-07-02:
Thanks @zelmon64 , reinstalling acat apparently solved the problem.

closed 2017-07-02 by gersonfs

zelmon64, 2017-07-04:
@gersonfs just to let you know, it seems like if you install new voices (as I just have) the icons will disappear again but reinstalling acat will bring them back again.

gersonfs, 2017-07-04:
Thanks @zelmon64 
It seems to me that the administrator of this project does not have the time to review these small bugs and the pull requests. They could pass the administration on to another developer.