intel/acat #55: Problem with selection of word of interest from predicted word list in Talk (Alt), Talk (ABC Alt), App (Alt) & App (ABC Alt) applications

ngidwan2, 2017-12-10:
There seems to be a problem/bug when trying to select a word from the "predicted word list" in the Talk (Alt), Talk (ABC Alt), App (Alt) & App (ABC Alt) applications using the keyboard or vision actuator (cheek or eyebrow gesture). Instead of selecting the word of interest, the next word in the word prediction list gets selected. Please note that the alphabets from the keyboard are selected correctly. 

A work-around (without changing code) is to select the word predicted previous to the word of interest, but if the word of interest is the first in the list after the undo button, one has to type the word completely, which is quite frustrating. Can anyone figure out what needs to be corrected in the source code and how to implement the change/correction in code in the above-mentioned applications? 

Thank you for your help.

zelmon64, 2017-12-10:
It's not the source code which is the problem but the xml configuration files for those modes (which is good because it means that the code doesn't need to be recompiled for it to work). The necessary corrections are in PR #49. You can download the files from that PR and replace the ones in your installation and the selection will work as it should.

ngidwan2, 2017-12-10:
Thank you @zelmon64. It worked!

closed 2024-11-19 by csavur