intel/acat #64: DUTCH Language Pack

alainverspecht, 2018-04-06:
I tried to reach Mr Prasad ( @saiprasadb01 ), up until now without success. 

Let me briefly introduce myself: my name is Alain Verspecht, 49 years old, living in Belgium, suffering from ALS since 2006. Founder of ‘Een hart voor ALS’ foundation, a Belgian non-profit organization raising funds for ALS Research. At this time being I can hardly move my head and I 'm losing my ability to speak. 

I have implemented ACAT on my home computer in the English language. I found the following on the ACAT page regarding new language packs:

> "Identify source material to create the word prediction database. The source should be free of copyright or other violations. Send us a link to the source."

After some research I found a link to Subtlex-nl, a joint project of the University of Ghent (Belgium) and the University of Tilburg (The Netherlands), resulting in a huge word database. You can have a first impression using the following link: 

http://crr.ugent.be/programs-data/subtitle-frequencies/subtlex-nl

Meanwhile I contacted one of the two leading scientists of this project, explained him what it is all about. He confirmed that he is willing to be our primary contact and that he is able to hand us over the most recent files available. 

So what I would like to hear from you, is whether it is possible to initiate this ACAT Dutch project and who will be our primary contact on your side? 

Hoping to hear from you very soon, 

Kind regards, 
Alain Verspecht

alexandre-mbm, 2018-04-09:
Hello, @alainverspecht! Hint: see also OptiKey and Tobii trackers. My father has used an EyeX with sucessful and his ALS is in a more advanced stage.

Galami, 2018-09-20:
Hello @alainverspecht,

Sorry for not answering earlier. I wanted to know if you found someone to do a Dutch language pack for ACAT. Let me also briefly introduce myself:

My name is Magali. I live in Brussels. I am a student who just finished my master's thesis on the configuration and evaluation by a user with cerebral palsy of the ACAT and Presage systems (the word predictor) in French (https://github.com/Galami/WordPrediction_AAC). For my thesis, I created a database in French. I looked at your link and I think maybe I could help you design the database in Dutch if you (and your association) are still interested, send me the most recent files available at mnorre@ulb.ac.be. We could discuss about several projects.

Kind regards,

Magali

sanscorp1, 2018-12-05:
I have never looked at this before but I want to help create a Dutch database.

It might be usefull to contact the e-mail address on the main github page for more info.

I'd like to know how this works and what I can do to contribute.

sanscorp1, 2018-12-21:
Still struggling with the program, did manage to create a database based on a free e-book and the link mentioned above. Manually removed as much garbage troughout the whole document (more then 20.000 words) but the database is fairly random in it's prediction. Used a 5 gram DB, will try a 3 gram DB.
Still searching how to translate the applications. I have all messages translated (resx) but the menu's and buttons are still in English.

Absolutely no experience with programming, learning as we go.

Help is welcome!

sanscorp1, 2019-05-04:
Never got any comment. My father is so bad he cannot use it any more.
I uploaded my work so far to Google Drive. Contains the Dutch Language Pack I created, also source files.
For someone else to improve on, I give up.

https://drive.google.com/open?id=1ZT_bK5zWaRuVQdx_m1MH4DJrExJKs48Y

closed 2024-11-19 by csavur