intel/acat #65: ACAT projects and feedbacks

JaneWayOnBoard, 2018-09-12:
Hello,
I am doing some researches over ACAT and I would like to know where I could find some user's feedbacks and projects related.

Many thanks,

Nathalie 
(Belgium)

Galami, 2018-09-13:
Hello, I just finished my master's thesis on the configuration and evaluation by a user with cerebral palsy of the ACAT and Presage systems (the word predictor) in French. I can send it to you if you want as well as the link to my repository github where there are some of my configuration files, photos and video of the user test (https://github.com/Galami/WordPrediction_AAC) . There is not much research on ACAT, I recommend the user guides (https://01.org/acat/documentation-list/) and the article: "Denman, P., Nachman, L., & Prasad, S. (2016, August), Stephen Hawking's UI, In Proceedings of the 14th Participatory Design Conference: Short Papers, Interactive Exhibitions, Workshops-Volume 2 (pp. 94-95). ACM ".

Magali
(Belgium)

JaneWayOnBoard, 2018-09-13:
Many Thanks !!

closed 2024-11-19 by csavur